greiman/UsbFat #2: initUSB() error

rosn, 2015-05-12:
When i try to execute initUsb function my Arduino pro mini crashes. In fact the real cause seems to be from usb->Init() where the function couldn't be resolved. You should take a look at it.

Thanks

greiman, 2015-05-12:
Which USB host shield are you using?

rosn, 2015-05-12:
I'am using the pro mini one. In fact i continued to search for the problem and it wasn't Init but Task() which kill the process and reboot the arduino

greiman, 2015-05-12:
The pro mini does not support USB host mode.  Are you using the Circuits@Home host shield?

rosn, 2015-05-12:
Yes I do.

rosn, 2015-05-12:
I found out why the board was reseting.
In Usb.cpp file, devConfig[] was declared but not initialize at NULL. So when in Task() you actualy called devConfig[i]->Poll() it went wrong and process was reseted. Maybe I did something wrong at init but you should look at it.

greiman, 2015-05-12:
This constructor initializes devConfig[]. It calls RegisterDeviceClass().

BulkOnly bulk(&usb);

See the UsbDriveInfo example.  It has these two lines

// USB host objects.
USB usb;
BulkOnly bulk(&usb);

I can't help you much more with the USB_Host_Shield library since I didn't write it.

greiman, 2015-05-12:
I setup my pro mini 3.3V with the Circuits@Home mini host shield and ran the UsbDriveInfo example.

Here is the result with a 32 GB Voyager

UsbFat version: 20141220
FreeRam 981
Type any character to begin
USB initialized.
Volume initialized.

Volume Size: 30992.92 MB (MB = 1,000,000 bytes)
Volume is FAT32
blocksPerCluster: 64
clusterCount: 945829
fatStartBlock: 18176
fatCount: 2
blocksPerFat: 7392
rootDirStart: 2
dataStartBlock: 32960
Determining free cluster count -  please wait.
freeClusters: 945828
freeSpace: 30992.89 MB (MB = 1,000,000 bytes)

Done

rosn, 2015-05-13:
"This constructor initializes devConfig[]. It calls RegisterDeviceClass()."

In fact it does init the array but not the values contains by the array. So sometimes it contains over value than NULL so RegisterDeviceClass() write the correct adress in a wrong position since first value different from NULL. Then when Task() and the prog tries to call Poll() in the first element it crashes. If you can contact the people in charge.

On an other hand I have a 5V Arduino pro mini but powered by a 3.3V source not to burn the shield. Last prog I tested was the logger data one. I'll try with the one you just told me to.

rosn, 2015-05-13:
Ok I made it work in my config. Thak you for your help. Just if you can contact the people in charge to correct the problem in the contructor.

closed 2015-05-13 by rosn