greiman/UsbFat #3: key.begin fails

AMORel69, 2016-06-20:
key.begin() fails whatever code example is run.
enable to find any suggestion in the documentation

greiman, 2016-06-21:
I just checked the examples and they work on my hardware.

Here is the bench on a Mega ADK
> Use a freshly formatted volume for best performance.
> 
> Type any character to start
> Free RAM: 6891
> Type is FAT32
> Volume size: 7.85 GB (GB = 1E9 bytes)
> File size 5 MB
> Buffer size 100 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 109.04,383216,88,911
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 157.91,5620,84,627

Here is the UsbDriveInfo example on an Uno with a Circuits@Home USB shield.

> UsbFat version: 20141220
> FreeRam 981
> Type any character to begin
> USB initialized.
> Volume initialized.
> 
> Volume Size: 7845.25 MB (MB = 1,000,000 bytes)
> Volume is FAT32
> blocksPerCluster: 32
> clusterCount: 478836
> fatStartBlock: 70
> fatCount: 2
> blocksPerFat: 3741
> rootDirStart: 2
> dataStartBlock: 7552
> Determining free cluster count -  please wait.
> freeClusters: 448983
> freeSpace: 7356.14 MB (MB = 1,000,000 bytes)
> 
> Done

I can't help you more since I don't debug user hardware.

closed 2016-06-21 by greiman

greiman, 2019-01-18:
Can't help you since I don't have this shield.

I will not be updating this library since there are not many users.

ThomasHenryLyly, 2019-01-24:
hi, I just change my board. I use Arduino ADK and use USBFat lib. But, when I plug my usb and other USBs, Serial Monitor said that: Error: initUSB failed. Please, help me

ThomasHenryLyly, 2019-01-24:
i have a question , when i use your USBFat lib, what is "Key"  ??
-----------------------------------------
Người gửi: Nguyễn Minh Tuấn
Email: bollbat5566@gmail.com
SĐT: 0916010298

greiman, 2019-01-24:
UsbFat supports USB flash drives formatted FAT32.

I tested an ADK with a 4 GB USB key.  I used Arduino IDE 1.8.8

Here are results from the UsbDriveInfo example.

> UsbFat version: 20141220
> FreeRam 7129
> Type any character to begin
> USB initialized.
> Volume initialized.
> 
> Volume Size: 4002.53 MB (MB = 1,000,000 bytes)
> Volume is FAT32
> blocksPerCluster: 32
> clusterCount: 244295
> fatStartBlock: 54
> fatCount: 2
> blocksPerFat: 1909
> rootDirStart: 2
> dataStartBlock: 3872
> Determining free cluster count -  please wait.
> freeClusters: 244026
> freeSpace: 3998.12 MB (MB = 1,000,000 bytes)
> 
> Done

I can't help you debug your setup and am no longer maintaining UsbFat.  There will be no future releases.