greiman/UsbFat #4: Disconnect and reconnect usb stick

fmder, 2016-06-23:
I was wondering if it was possible anyhow to disconnect and reconnect the usb stick from the USB Host Shield without restarting the arduino.

I tried putting the initialization in the loop, but I'm unable to open a file after the usb stick is disconnected and reconnected.

greiman, 2016-06-23:
Something in the USB_Host_Shield library fails when initialization is done a second time.

I didn't write this library so I don't know why.  It may be necessary to call a function in this library to allow removal of the usb stick.

I don't plan future development of UsbFat since the [USB_Host_Shield](https://github.com/felis/USB_Host_Shield_2.0) library has marginal USB mass storage support.

closed 2016-06-23 by greiman

didix21, 2016-09-14:
I have done some proves and when I disconnect and recoonet the usb stick the error which I obtain is from bulkonly not from USB class. Then, are you sure that the problem is on USB_Host_Shield ?

greiman, 2016-09-14:
The BulkOnly class is in the USB_Host_Shield_2.0 library.  You will need to explore this since I didn't write the library.

You should start with a new version of the USB Host shield library since the version included with UsbFat is very old now.