greiman/UsbFat #7: Usbfat incompatible with SD.h or sdfat.h

shadeyinka, 2017-07-26:
Thanks for your excellent USB host code for data storage.

I am working on a project which uses a Mega ADK Arduino with an SD card  shield. The project is basically a data logger with primary storage as the SD card while the secondary is the USB flash drive.

The USB flash drive is to download the files on the SD card.

I can log on SD card separately
AND
Use your excellent USB code separately
BUT
There seems to be a clash when sdfat.h or SD.h is added to your code. 

Is there a way out? I need to use be able to copy files on the SD card to the flash drive. Thanks

greiman, 2017-07-26:
UsbFat uses code common to SD.h and SdFat.  It would be possible to support USB sticks and SD cards with a modern version SdFat but there has been little interest in UsbFat so I have not worked on it.

shadeyinka, 2017-07-29:
Thanks so much for your support. I did a further investigation and I found
out that Usbfat.h and sdfat.h are compatible and SD.h isn't compatible with
Usbfat.h.

However the only problem is that incompatibility issues arise with sdfat.h
with Usbfat.h when this cast is used

Sdfat SD;

I will try to see how I can use pure code with Sdfat to copy files into USB
memory device.

Thanks so much

Yinka

shadeyinka, 2017-08-02:
I am still stuck Sir. I do not know what I am doing wrong. I just want to use your USB host with Flash Drive with SD Card. Nothing seems to make the program compile.
SD Card program works separately and USB mass storage works separately.

Any help will be appreciated.

Yinka

greiman, 2017-08-02:
It takes a major restructuring of the code.  I don't plan to do it since there is little interest in UsbFat.

shadeyinka, 2017-08-02:
OK. Thanks a lot. It looks like its going to be a tedious revision. I hope
in future, you'll look into that.

I appreciate you.

shadeyinka, 2017-09-12:
I attempted to merge the common codes in the UsbFat Library and the SdFat Library
1. I created a new folder "FatLib" in the Library folder
2. I copied the common files from
    ..\SdFat\src\FatLib\     in SdFat Library and
    ..\UsbFat\UsbFat\utility   in UsbFat Library
    inclusive of the File: BaseBlockDriver.h
3. In  ..\SdFat\src\  
 a.    I edited     #include "FatLib/FatLib.h" to "FatLib.h" 
 b.    I edited    #include "FatLib/BaseBlockDriver.h" to "BaseBlockDriver.h"
4. In.  ..\UsbFat\
a.      I edited     #include "utility/FatLib.h" to "FatLib.h" 
5. I  then added 
 a.   #include <FatLib.h> to the example file ReadWrite.ino
and
 b.  #include <FatLib.h>  to the example File UsbLogger.into
6.  All the example Files of the SdFat Library Compiles and Executes
     However, the Example Files of UsbFat gives Compiler errors 
     error: invalid conversion from 'uint8_t {aka unsigned char} to 'BlockDriver* {aka SdSpiCard*}' [-permissive].

Any hint on how to resolve this? 
Thanks for your understanding. I will post both my library and the error after this.

shadeyinka, 2017-09-12:
My Errors: I use Arduino IDE 1.8.3, AVR Board manager 1.6.11

In file included from C:\Users\HP\AppData\Local\Temp\arduino_modified_sketch_727076\UsbLogger.ino:6:0:

C:\Users\HP\Documents\Arduino\libraries\UsbFat/UsbFat.h: In member function 'bool UsbFat::begin(uint8_t, uint8_t)':

C:\Users\HP\Documents\Arduino\libraries\UsbFat/UsbFat.h:62:37: error: invalid conversion from 'uint8_t {aka unsigned char}' to 'BlockDriver* {aka SdSpiCard*}' [-fpermissive]

     return FatFileSystem::begin(part);

                                     ^

In file included from C:\Users\HP\Documents\Arduino\libraries\FatLib/FatLib.h:24:0,

                 from C:\Users\HP\AppData\Local\Temp\arduino_modified_sketch_727076\UsbLogger.ino:4:

C:\Users\HP\Documents\Arduino\libraries\FatLib/FatFileSystem.h:44:8: error:   initializing argument 1 of 'bool FatFileSystem::begin(BlockDriver*, uint8_t)' [-fpermissive]

   bool begin(BlockDriver* blockDev, uint8_t part = 0) {

        ^

exit status 1
Error compiling for board Arduino Mega ADK.

shadeyinka, 2017-09-12:
My Library is attached here

[UsbFat_SdFat Library.zip](https://github.com/greiman/UsbFat/files/1295146/UsbFat_SdFat.Library.zip)

Thanks.

From
red and sore eyed Yinka

greiman, 2017-09-12:
I can't help you.  I have no interest in merging the current SdFat code with the old UsbFat library.  

I am in the process of developing a totally new library that supports FAT12/FAT16/FAT32 and exFAT.

I only plan bug fixes for SdFat.

shadeyinka, 2017-09-12:
Ok.
Thanks...with tears in my eyes

vladkozlov69, 2019-03-22:
@shadeyinka if it's still  actual, I have the modified version which seems to be compatible with OOTB SD library

shadeyinka, 2019-03-22:
Thank you so much.
I got the last one to work by modifying the sdfat.h file.

I'll check it out

Thanks

imranaalam, 2019-10-16:
@shadeyinka can you share the updated code. I want to do the same as you I.e. mixing sdcard and usb. 
Or any guidelines on how to achieve it?

imranaalam, 2019-10-16:
@greiman when can we expect the new library. We are anxiously waiting for it?

vladkozlov69, 2019-10-16:
> @shadeyinka can you share the updated code. I want to do the same as you I.e. mixing sdcard and usb.
> Or any guidelines on how to achieve it?

You can see what I did in my fork at https://github.com/vladkozlov69/UsbFat/commit/6d88d2bcb4add5428ac2939a4c3f0cd9ded3b28b maybe it helps

greiman, 2019-10-16:
I have a working example using a USB drive with SdFat-beta but have not had time to add it to the library.  I need to make SdFat-beta the release version of SdFat first.

imranaalam, 2019-10-21:
@greiman 
thanks. there is a hope. we are hopeful that you will get the time this weekend.

greiman, 2022-07-06:
I have not tried the code in almost two years.  I used this [old usb library](https://github.com/greiman/SdFat/tree/master/extras/USB_Host_Shield) with SdFat.  I didn't use any code from the UsbFat library.

Here is the sketch but it may not work with the latest SdFat and I have given up on keeping it working with the USB_HOST_SHIELD library.

[UsbKey.zip](https://github.com/greiman/UsbFat/files/9055236/UsbKey.zip)

So I don't have the time or interest to get this code working again.

proasnet, 2025-08-27:
Dear @greiman , I am returning in the time. I have a few years old project, where I have an SD and USB on the board with copy file from SD to USB. Two years later, all working. I not have updated libraries from this time (2023).
Now, If I compile and flash edited project, USB after init return code
```
SD karta inicializována
Host initialized, ms: 1003
USB powered, ms: 1004
USB state: 20, ms: 1004
USB state: 40, ms: 1204
USB state: 50, ms: 1254
USB state: 51, ms: 1254
USB state: A0, ms: 1274
```
This log I get, If I init SD card in the same sketch.

If I Init USB only, USB was init fully. The last state is 90.
Please, have you any idea, where is a problem?
I understand that you no longer support this library, but I have no choice but to use your library.
Thank you very much for your time.

```
USB usb;
BulkOnly bulk( &usb );
UsbMscDriver usbKey( &bulk );
FsVolume key;

SdFat sd;

void setup() {
  Serial.begin( 115200 );
  DDRK |= (1 << PK4); // O PK4 SD ss
  Reset_PK4;

  
  //----------------------------------------------------------------------------------------------------

  if (!sd.begin( SD_CONFIG )) {
    Serial.println( "Inicializace SD karty selhala" );
  }
  Serial.println( "SD karta inicializována" );

  //----------------------------------------------------------------------------------------------------
  if ( !initUSB( &usb ) ) {
    Serial.println( "Chyba: Inicializacia USB" );
  } else Serial.println( "Inicializacia USB uspesna" );

  if (!key.begin(&usbKey)) {
    Serial.println( "Chyba: Start klice" );
  } else Serial.println( "Start klice" );
  //----------------------------------------------------------------------------------------------------

}

void loop() {}
```