greiman/UsbFat #8: GPLv3 Compatibility

marcio-ao, 2018-03-15:
Hello,

I work for [LulzBot](https://www.lulzbot.com/), a maker of fully open-source 3D printers. We would like to incorporate code from the USB Host Shield into the [Marlin firmware](http://marlinfw.org/) for 3D printers. This would allow printers to read USB Flash Drives rather the SD cards. The company I work for is a great believer in open-source and we would donate all our integration work to the open-source and 3D printing community.

However, the USB Host Shield was released under "GPL v2" rather than "GPLv2 or later", which would preclude integration with Marlin, which is released under the "GPL v3".

I have already [reached out to the developers of that library](https://github.com/felis/USB_Host_Shield_2.0/issues/364#issuecomment-373245669) and they have agreed to change their license from "GPL v2" to "GPL v2 or later" in order to resolve this issue. The reason why this is necessary is listed [here](https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility).

I am reaching out to you because I would like to be able to include the modifications you have made that allow USB flash drives to mount faster. For me to be able to do so, I would like to know whether you would be on board with the license change from "GPL v2" to "GPL v2 or later".

Please refer to the original thread here for the discussions related to this:

https://github.com/felis/USB_Host_Shield_2.0/issues/364#issuecomment-373245669

Thank you!

-- Marcio

greiman, 2018-03-17:
I have changed the license for SdFat to MIT.  UsbFat is based on SdFat so I would be happy to change to MIT.
 
The MIT license is more permissive than GPL so would that satisfy your needs?
 
```
/**
* Copyright (c) 20011-2017 Bill Greiman
* This file is part of the SdFat library for SD memory cards.
*
* MIT License
*
* Permission is hereby granted, free of charge, to any person obtaining a
* copy of this software and associated documentation files (the "Software"),
* to deal in the Software without restriction, including without limitation
* the rights to use, copy, modify, merge, publish, distribute, sublicense,
* and/or sell copies of the Software, and to permit persons to whom the
* Software is furnished to do so, subject to the following conditions:
*
* The above copyright notice and this permission notice shall be included
* in all copies or substantial portions of the Software.
*
* THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS
* OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
* FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
* AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
* LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING
* FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
* DEALINGS IN THE SOFTWARE.
*/

```

marcio-ao, 2018-03-19:
UsbFat includes code from USB Host Shield, so it would have to carry the same license. Your options are to do nothing and keep your contributions under GPLv2 (your code would become a fork USB Host Shield before the licence change), or go along with the changes in upstream to "GPLv2 or later".

As you are willing to license under the MIT, I believe your best choice for USBFat is to go to "GPLv2 or later". This is what would allow me to integrate your contributions.

Thank you,

-- Marcio