greiman/UsbFat #9: Error key.begin

saikiran7714, 2018-07-26:
sir im using your usbFat library with usb host shield but getting key.begin error . In usb host library i can get my drive information

saikiran7714, 2018-07-26:
[image]
please help me  sir....

greiman, 2018-07-26:
I can't help.  I am no longer updating this library since it was not popular.

greiman, 2018-07-26:
I did try the UsbDriveInfo example using an Aduino Maga ADK with a SanDisk 8 GB Cruzer Blade key.

It seems to work.

> UsbFat version: 20141220
> FreeRam 7132
> Type any character to begin
> USB initialized.
> Volume initialized.
> 
> Volume Size: 7844.90 MB (MB = 1,000,000 bytes)
> Volume is FAT32
> blocksPerCluster: 32
> clusterCount: 478815
> fatStartBlock: 742
> fatCount: 2
> blocksPerFat: 3741
> rootDirStart: 2
> dataStartBlock: 8224
> Determining free cluster count -  please wait.
> freeClusters: 471365
> freeSpace: 7722.84 MB (MB = 1,000,000 bytes)
> 
> Done

closed 2018-07-27 by saikiran7714