greiman/UsbFat #10: UsbFatBench: key.begin fails on second loop

vladkozlov69, 2019-03-22:
Hi Bill,

I'm just playing with UsbFatBench and it works just fine on the first loop() execution, but when it again prompts "Type any character to start" on a 2nd loop execution, key.begin fails:

> Type any character to start
> 
> Type is FAT32
> Volume size: 15.50 GB (GB = 1E9 bytes)
> File size 5 MB
> Buffer size 100 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 152.27,95995,1,655
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 192.75,5001,1,517
> 
> Done
> Type any character to start
> 
> key.begin failed

I know that you are not developing this project anymore, but maybe you can give me some hints where to look and what could be the root cause of this behaviour? Thanks in advance.

Meins321, 2020-10-27:
Atleast your demo worked once..
i only get "key.begin failed" with all the examples :-/

i will try to run it

Meins321, 2020-10-27:
15:44:18.405 -> FreeRam 845
15:44:18.405 -> Type any character to begin
15:44:21.013 -> FreeRam 845
15:44:21.013 -> Type any character to begin
15:44:27.105 -> Host initialized, ms: 1003
15:44:27.105 -> USB powered, ms: 1003
15:44:27.139 -> USB state: 20, ms: 1007
15:44:27.274 -> USB state: 40, ms: 1203
15:44:27.341 -> USB state: 50, ms: 1253
15:44:27.341 -> USB state: 51, ms: 1253
15:44:31.810 -> USB state: 90, ms: 5731
15:44:31.810 -> key.begin failed

greiman, 2020-10-27:
I have had given up support for this library since it requires the [USB_Host_Shield_2.0](https://github.com/felis/USB_Host_Shield_2.0) library and new versions of this library no longer work with most USB keys.

I don't want to look for the problem in the 2014 version of USB_Host_Shield_2.0 included with UsbFat.