greiman/UsbFat #11: Support - USB Host Shield - MAX3421E - Arduino STM32 (STM32F103C8T6, Bluepill)

rtek1000, 2019-11-04:
I tried to compile with bluepill (STM32F103CbT6), but because this uC already has internal USB, conflict occurred. So I had to change the library to compile.

If anyone wants to test to see if it works, follow attached.

https://github.com/stm32duino/Arduino_Core_STM32

[UsbFat_STM32F103-master.zip](https://github.com/greiman/UsbFat/files/3803228/UsbFat_STM32F103-master.zip)

xxxajk, 2019-11-04:
While not fully supported yet, can you check to see if there are any conflicts in UHS30?

rtek1000, 2019-11-04:
I'll try, thank you.

xxxajk, 2019-11-04:
It might work out-of-the-box for the max chip *IF* whatever you are using actually follows a standard. You MIGHT have to teach it a new board, but shouldn't have to. Since it is ARM, most of the important stuff is already taken care of for you.

rtek1000, 2019-11-05:
Yes, it's out of the box, but I found it pretty weird how the 2.0 library works with pins. The Arduino platform is being extended to the out-of-box boards, but the pins were still declared again in the library.

closed 2019-11-05 by rtek1000