greiman/UsbFat #13: compilation error in UsbFatDemo example

hody00, 2020-11-18:
hello :)

I tried to run UsbFatDemo example and got this compilation error:

```
Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), board:"Arduino Mega ADK"

C:\Users\hoday\AppData\Local\Temp\ccOlzTjz.ltrans0.ltrans.o: In function `ls':

sketch/FatFileSystem.h:185: undefined reference to `FatFile::ls(Print*, unsigned char, unsigned char)'

C:\Users\hoday\AppData\Local\Temp\ccOlzTjz.ltrans0.ltrans.o: In function `open':

sketch/FatFile.cpp:395: undefined reference to `FatFile::parsePathName(char const*, fname_t*, char const**)'

sketch/FatFile.cpp:402: undefined reference to `FatFile::open(FatFile*, fname_t*, unsigned char)'

sketch/FatFile.cpp:410: undefined reference to `FatFile::open(FatFile*, fname_t*, unsigned char)'

collect2.exe: error: ld returned 1 exit status

exit status 1

```

it seems like several functions were declared but not implemented.

I cant just delete declerations because those functions are called inside the code.. 

Any advice?

greiman, 2020-11-19:
This repository contains two libraries.  You must install the USB_Host_Shield_2.0 and UsbFat folders in your libraries folder.  Don't copy the UsbFat-master folder, just copy the above folders.

It is six years old but still seems to compile and work on a Mega ADK.  But I am no longer maintaining it since new versions of USB_Host_Shield_2.0 library fail.  You must fix any bug yourself.

Here is the output from the UsbFatDemo example today with the 1.18.3 Arduino IDE.

```
FreeRam 7042
Type any character to begin

Volume Size: 8270 MB

2015-08-15 08:00:08          0 System Volume Information/
1980-00-00 00:00:00         11 usbtest.txt
2000-01-01 01:00:00         11 test file.txt
2015-08-22 09:54:48    3255999 Intel_Chipset_Win10_V10_0_27.zip
2015-08-22 09:54:16   13600728 Realtek_LAN_Win10_V1015052015.zip
2015-08-22 09:55:50          0 Intel_Chipset_Win10_V10_0_27/
2015-08-22 09:56:00          0 Realtek_LAN_Win10_V1015052015/

Done
```

hody00, 2020-11-19:
you are right, turns out I didn't include all the relevant libraries..
It works! Thanks :)

[image]

closed 2020-11-19 by hody00