greiman/UsbFat #18: How to reinitialize usb device?

myslivec, 2023-07-14:
Hi,
is there any way, how to reinitialize the USB device. On the start of program I initialize USB device, creat file, write file and close file, then I would like to reinitialize it, for subsequent connection of a new device
Does there exist any file with functions and key words for library UsbFat?

Thanks. Tomas

greiman, 2023-07-14:
The UsbFat uses an old unsupported version of SdFat.  With SD card it was possible to just call begin() again.

For USB the USB_Host_Shield_2.0 library I used would need to be reinitialized.  I don't know if that is possible since I didn't write it.

It is possible to use the new version of SdFat with USB devices.  I did a test example but there is so little request for USB support that I didn't update this library.  USB_Host_Shield_2.0 keeps breaking for USB mass storage.

I attached the example and it works with with the new SdFat and the version of USB_Host_Shield_2.0 I have.  It has this patch in masstorage.cpp

```
uint8_t BulkOnly::Page3F(uint8_t lun) {
        uint8_t buf[192];
        for(int i = 0; i < 192; i++) {
                buf[i] = 0x00;
        }
        WriteOk[lun] = true;
        return 0;  // WHG - Many USB keys don't respond. <<<<<<<<<<<<<<<<<<<<<<< Added
        uint8_t rc = ModeSense6(lun, 0, 0x3f, 0, 192, buf);
        if(!rc) {
                WriteOk[lun] = ((buf[2] & 0x80) == 0);
                Notify(PSTR("Mode Sense: "), 0x80);
                for(int i = 0; i < 4; i++) {
                        D_PrintHex<uint8_t > (buf[i], 0x80);
                        Notify(PSTR(" "), 0x80);
                }
                Notify(PSTR("\r\n"), 0x80);
        }
        return rc;
}
```

> Here is output from the example:
> Type any character to start
> Host initialized, ms: 1004
> USB powered, ms: 1004
> USB state: 20, ms: 1009
> USB state: 40, ms: 1204
> USB state: 50, ms: 1254
> USB state: 51, ms: 1254
> USB state: 90, ms: 6702
> 2023-01-01 00:00         22 usbtest.txt
> 2018-09-15 00:28    1004054 img102.jpg

[UsbKey.zip](https://github.com/greiman/UsbFat/files/12050524/UsbKey.zip)

myslivec, 2023-07-17:
Thank you for your fast answer. 
I found non-standard solution - software restart processor and initialize again. But this solution is non-possible in many situations. I will try to use added example and library USB_Host_Shield_2.0.

proasnet, 2024-01-30:
Is there anything new in this solution? I use UsbFat on my device, to export data from the microSD, it copies the csv files from the card. If I unplug the USB and plug it back in, it won't initialize again. A re-initialization adjustment would help.