greiman/UsbFat #19: fat 4gb usb stick "initUSB Failed"

allegfede, 2025-02-05:
UsbFatDemo untouched and a fat usb stick with only 1 text file inside.
How to check what's wrong?

greiman, 2025-02-05:
This library is unsupported and not been modified in eleven years.  I found many USB sticks would not initialize with the USB_Host_Shield_Library_2.0 library.

I tried the using the current version of [SdFat](https://github.com/greiman/SdFat) as described [here](https://github.com/greiman/UsbFat/issues/18#issuecomment-1635820372) with the current unmodified version of [USB_Host_Shield_Library_2.0 ](https://github.com/greiman/UsbFat/tree/master/USB_Host_Shield_2.0)library and had success with several USB keys.

Here is the output for a ADATA 32GB key with a USB host shield on an Arduino Mega2560.

> Type any character to start
> Host initialized, ms: 1003
> USB powered, ms: 1004
> USB state: 20, ms: 1008
> USB state: 40, ms: 1204
> USB state: 50, ms: 1254
> USB state: 51, ms: 1254
> USB state: 90, ms: 6883
> 2025-01-01 00:00         11 usbtest.txt

I can't help diagnose problems with USB_Host_Shield_Library_2.0 library.

allegfede, 2025-02-05:
thanks. You are very kind answering :D

allegfede, 2025-02-07:
by the way, i soldered the 3.3v and the 5v vin pads and the 5v vout pad and i can now run the UsbFatDemo code u provided 😄 

Now i need to find how to read a text file i got on the usb stick ;-)

thanks